tcunit/TcUnit #186: Curious error: variable "Test" prevents stop at a breakpoint

Pippen19, 2022-06-03:
If I instantiate a variable with the name "Test" and the type "UINT" in an FB derived from FB_TestSuite, then the program no longer stops there at a breakpoint.

If there is no variable "Test" in this method, the program stops at the breakpoint:
[image]

If there is a variable "Test" in this method, the program does not stop at the breakpoint:
[image]

On the fly, I haven't figured out what triggers the problem. Maybe it is also a Beckhoff problem (even probable). But it is astonishing, because it only occurs in FBs that are derived from FB_TestSuite.

If I use STRING as type instead of UINT the following error occurs when trying to set a breakpoint in this method:
[image]

sagatowski, 2022-06-03:
This looks weird indeed.
Did you declare the variables as VAR or VAR_INST? Is there a difference?
Which version of the applicable software did you use? (TcUnit library, TwinCAT XAE version, Visual Studio version etc)
Under what environment did you run the software? PLC? Virtual machine?
If you rename the variable to something else than Test, does this still prevail?
If no, does the varialbe have to be named Test?

Pippen19, 2022-06-03:
No, there is no difference between VAR or VAR_INST.
I use TcUnit library.
I use it on a PLC.
No, I have the problem only with the name "Test".

In the meantime I found out that it has nothing to do with TcUnit, but is a fundamental problem: The problem is that the program in which the FB is instantiated is also called "Test". Apparently the namespaces are not cleanly separated. I have reported it to Beckhoff.

sagatowski, 2022-06-03:
I was more curious of what **versions** of TwinCAT, TcUnit, etc you were using to reproduce the problem.

But if you know the problem is not related to TcUnit, I will close this issue.

closed 2022-06-03 by sagatowski

PTKu, 2022-06-03:
This problem occurs on other CoDeSys systems as well. It's hard to reproduce but my guess would be there is a problem between Online (PLC program) and Editor's debug information.

Pippen19, 2022-06-03:
What do you mean with "hard to reproduce"? I find it's easy to reproduce (see my little dummy project).

PTKu, 2022-06-03:
> What do you mean with "hard to reproduce"? I find it's easy to reproduce (see my little dummy project).

More complex the program, more likely you will encounter this problem, it those scenarios is it hard to pinpoint where exactly the problem is, this is what I meant. I will have a look at your project, thanks.

PTKu, 2022-06-03:
🤣 just did try... this problem is even more silly than the one I had in mind. Thanks @Pippen19.