tcunit/TcUnit #187: Variable "NumberOfTestsToAnalyse" is limited to the wrong boundaries

JoergWitt, 2022-06-07:
**Describe the bug**
It is not possible to set the GVL variable `MaxNumberOfTestSuites` less than `MaxNumberOfTestsForEachTestSuite`. It causes compiler errors in `FB_TestResults` and `FB_TestSuite`.

The reason is the variable `NumberOfTestsToAnalyse` used in TcUnit. It is limited to:
`NumberOfTestsToAnalyse : UINT(1..GVL_Param_TcUnit.MaxNumberOfTestSuites);`

The solution is as follows, if you agree my **Expected behavior**:
`NumberOfTestsToAnalyse : UINT(1..GVL_Param_TcUnit.MaxNumberOfTestsForEachTestSuite);`

**Expected behavior**
I think, that number of test suites and number of test in each test suite are independent from each other and so I can set them independently.

**Screenshots**
[image]

JoergWitt, 2022-06-13:
Sure, will create the pull request.

sagatowski, 2023-04-29:
Issue solved. Thanks @JoergWitt.

closed 2023-04-29 by sagatowski