tcunit/TcUnit #196: Verbose mode

Rednas83, 2022-10-23:
Would be cool to have a verbose mode which only displays failed tests in the output window.

sagatowski, 2023-04-29:
Isn't that the opposite of verbose to only show parts of the output?
Currently the only thing that is shown in the output window (in error mode) is the failed tests (or more correctly, the failed assertions) + tests results, so this is exactly what you ask for?

Rednas83, 2023-04-30:
This is indeed what I asked for. Verbose is indeed all messages and non-verbose only the essential ones. How can this error mode be activated?

sagatowski, 2023-04-30:
You have the essential ones in the ERROR-list of VS, while the non-essential ones are just normal messages in visual studio. 
[image]

You have to make sure "Messages" is selected.

Rednas83, 2023-04-30:
I don't like this approach since I also use Messages for debugging applications. Is it not possible to add a configuration option for this?
Something like:
[image]

sagatowski, 2023-04-30:
Sorry, I really don't understand what you are asking for.
Initially you only want failed tests to be shown, which is EXACTLY what you already have.

Rednas83, 2023-04-30:
Currently I am not working on a tc3 project. So I had to look at the docs for this.
Verbose mode=FAILED TESTS and STATISTICS
[image]
Non verbose mode= ONLY FAILED TESTS

Just a way for users to opt out if they are not interested in the statistics.

sagatowski, 2023-04-30:
I don't think there is a strong usecase for this. Normally when you write your tests (applying the [practices of TDD](https://www.youtube.com/watch?v=FnXY6MA3axw), you ALWAYS want to know that all the tests you've just written have failed, and the normal procedure is that you write a bunch of tests, and make sure that the count adds up. I think this is much easier than counting the number of asserts and making sure it adds up (especially since the amount of asserts can differ than the amounts of tests, as you can do more than one assert per test).

closed 2023-11-09 by sagatowski