tcunit/TcUnit #200: State of the project

stefanbesler, 2022-12-07:
How is the state of this project @sagatowski ?
I do have some ideas, but as pull requests are already lined up so it will take even more time to get my stuff in.
I can obviously just fork the repository, but I would only do this if you are no longer willing to maintain this repo and/or not give invites to be part of the tcunit organization.

sagatowski, 2022-12-07:
I've had bad conscience over not having any time to work on the pull requests and follow them up lately. I know there are a lot of users that need this. I've been receiving many e-mails lately with questions about TcUnit, and the state of it, even from very big companies. The usage of this framework is well beyond my imagination. I see it EVERYWHERE in the industry. And I get a lot of messages of users working in various industries asking questions about the framework. This of course makes me very happy, but at the same time...

All my spare-time over the last 2 years or so have gone into making the [free TwinCAT 3 PLC-programming course](https://www.youtube.com/playlist?list=PLimaF0nZKYHz3I3kFP4myaAYjmYk1SowO). This course has literally eaten up every minute of my spare-time, vacations, holidays and I even had to work less in my daily work to finish this course.

LITTERALY. EVERY. MINUTE.

As both TcUnit (and the TwinCAT 3 PLC programming course) is fully un-paid work (the other way around, it costs money to maintain both of them), I now have to put some priority on my normal work as I still have a rent to pay.

So the plan ahead?
As I just finished making the free TwinCAT 3 course, I first need to take some time off and rest. As soon as I've rested, I will work through all the pull requests & issues (and update the website too, lots of updates need to go in there). Once I've done this, I will think of what the next step will be (adding users to the organization, or setting another fork as master, or...?). There are some very big companies using TcUnit on a daily basis, which of several of them should be able to afford to add a maintainer or two. I will have to think about this.

If you are in a hurry, you are of course more than welcome to fork the project ❤️

stefanbesler, 2022-12-07:
Thanks, of course I understand, I have open source repos as well that I maintain completely unpaid or donated for.
Just wanted to know if you set this project aside and somebody has / should step in or if it is still on your radar. Seems not to be the case so thank you.

I think it would be benefical to open the tcunit organization by adding users and let them push branches and only protect the main branch by only letting you push to it via pull requests.

sagatowski, 2022-12-07:
What is the advantage/difference of having someone in the organization and push branches versus just doing pull requests through branches in their respective forks?