tcunit/TcUnit #202: Adding solution file path as a parameter for the TcUnit-Runner

ElliHerrmann, 2022-12-09:
**Description - New feature

Expected behaviour**

Jenkins Pipeline works with the right solution (Test- solution)

**Informations**

The TwinCat project has 2 solutions in 1 project. We use the TCUnit-Runner to check our code with a Jenkins Pipeline.

When we call the TcUnit-Runner we want to check the functionality of our code with our test solution. Right now TcUnit-Runner is trying to work with the productive code solution but we want to work with our test solution.

Is there already a way to order the Jenkins job to check a self chosen Solution (define the path to the solution in Jenkins) or is it only possible to find the correct solution by enhancing TCUnitRunner?

**Software- Version**

TCUnitRunner Version 0.9.3.0

iadonkey, 2022-12-18:
If you are using Jenkins and the 2 solutions are in separate folders, you could use dir(“folder_of_solution2”) {
  // execute tcunitrunner
 }

ElliHerrmann, 2023-01-17:
Hey iadonkey,

thank you for answering.
I will try your suggestion.