tcunit/TcUnit #205: FB_ReadJsonFile_Test

nico10hahn, 2023-01-04:
I want to use TC_Unit to check if a JSON File is written correctly by a function of my library. Therefore I want to create a test FB, which opens the written file and check if  the test string is written inside the file (for example timestamp).
I will parse the JSON Document with FB_JsonDomParser from TC3_JsonXml.

I tried different ways but doesn't work. Its possible to parse a document when I call the FB with TCUnit.Run() ?

When I usw my code in a normal FB I can read the timestamp.

sagatowski, 2023-01-04:
Yes, this is possible.
You open the file like you would normally do, and then simply assert that the content is what you expect.

nico10hahn, 2023-01-05:
Thanks for answer and the help,

to read the JSON file I use the code from the "Getting started" from Beckhoff Homepage. 

This is the main programm (just call the FB), for testing I added the whole code in the FB (later i want to use methods as in your youtube channel and the examples).

[image]

The FB with the TC_Unit interface looks like that:

[image]

-> It only logs the "I Arrive here:  Yes" message. 
-> When I use the same code and call the FB without TcUnit.Run() I get many logging messags with timestamp. 
-> It seems that "TcUnit.RUN()" calls the FB only onves even when I don't use "FINISHED()" in the FB. 

Thanks for help

closed 2023-01-09 by nico10hahn