tcunit/TcUnit #209: Assert does not play nice with properties

slabarque, 2023-03-20:
**Describe the bug**
Asserting the expected value of a property makes TcXaeShell crash and triggers a windows Blue-Screen-Of-Death causing a full pc reboot.

**To Reproduce**
Follow the [introduction user guide](https://tcunit.org/introduction-user-guide/). Then add a property to FB_Sum and assert it's value in a test.

**Expected behavior**
No crash.

**Software versions**
Which version of the applicable software did you use? 
- TcUnit library: 1.2.0.0
- TcXaeShell Version 15.0.28307.1913 D15.9
```
TcProjectCompare   1.0
Visual Studio Integrated Version of TcProjectCompare

TcTargetBrowserPackage Extension   1.0
TcTargetBrowserPackage Visual Studio Extension Detailed Info

TcXaeDebuggerLiveWatch   1.0
TwinCAT XAE Live Watch Tool Window

TcXaeHelper   4024.40.0.0
Info

TcXaeModules   4024.40.0.0
TwinCAT TMC Editor, TwinCAT TMC Code Generator

TcXaeShell   1.7.1
TwinCAT XAE Shell customization

TcXaeStartPage   1.13
Beckhoff Start Page for TwinCAT XAE

TwinCAT 3 AML Data Exchange    1.10.0.0
Provides import and export of AutomationML for TwinCAT

TwinCAT Bode Plot   3.4.3147.12
Integration package for TwinCAT Bode Plot components

TwinCAT Filter Designer   3.4.3147.12
Integration package for TwinCAT Filter Designer components

TwinCAT Measurement   3.4.3147.12
This package contains the project system for TwinCAT Measurement projects.

TwinCAT Scope   3.4.3147.12
Integration package for TwinCAT Scope components

TwinCAT XAE Base   3.1.0.0
The TwinCAT XAE Base extension provides the basic project structure for TwinCAT projects.

TwinCAT XAE EventLogger   4024.40.0.0
The TwinCAT XAE Eventlogger extension provides the capability of monitoring events sent through the TwinCAT 3 Eventlogger of a target system.

TwinCAT XAE PLC   3.1.0.0
PLC Environment
```

**Run environment**
Soft PLC

**Additional context**
I created a [repo](https://github.com/slabarque/TcUnitAssertPropertyIssue) that demonstrates the issue. The [fix](https://github.com/slabarque/TcUnitAssertPropertyIssue/tree/feature/fix_prop_assert_issue) is simple, work with an in between variable.

sagatowski, 2023-11-23:
I can confirm that you get a BSOD with TcUnit 1.2.0.0.
It's a more graceful crash using the current master of TcUnit (1.2.1.0), but it still crashes.

Can't quite figure out what the problem is, however...
As you correctly say, one fix is to use an inbetween variable, but another solution is to use Assert using the primitive type that you are asserting.
Another alternative is to use assertequals for the correct primitive type. I noticed that you are using `AssertEquals()`. If you use `AssertEquals_BOOL()` instead, it works just fine on properties.

sagatowski, 2023-11-24:
@slabarque 

This is according to the compiler. I've documented it in the FAQ here:
https://tcunit.org/#/faq?id=_13-assertequalsany-on-properties-makes-my-development-environment-crash-why

As this is not a limitation in TcUnit, I will close this ticket now.

closed 2023-11-24 by sagatowski