tcunit/TcUnit #210: Duration is not calculated when running tests in sequence

sagatowski, 2023-04-29:
**Describe the bug**
Whenever running tests in sequence (any of the examples [here](https://github.com/tcunit/ExampleProjects/tree/master/RunTestsInSequenceExampleProjects), the duration is always calculated to be 0.0. 

**To Reproduce**
Clone the latest master, run any of the examples above, and see the duration be 0.0.

**Expected behavior**
Duration should be correctly calculated, just like in the case when running tests that are not in sequence (so running in parallell).

**Screenshots**
[image]

**Software versions**
TwinCAT 3.1.4024.44.
Visual Studio 19 (according to the contributing guidelines).
TcUnit from master.

**Run environment**
Running in virtual machine.

**Additional context**
Note that build-chain for TcUnit has been updated. Please look at Contributing markdown file.

sagatowski, 2023-04-29:
@stefanbesler I just merged your changes in. Unfortunately, not everything works. See above.

sagatowski, 2023-05-01:
Initial fix done. Needs to be tested more. See #211 for the details.

sagatowski, 2023-11-19:
Fixed.

stefanbesler, 2023-11-24:
Hm, not sure what went wrong here. What’s left to do, where can I help?

sagatowski, 2023-11-26:
Hey! I ran all the [timed/sequence examples](https://github.com/tcunit/ExampleProjects/tree/master/RunTestsInSequenceExampleProjects) to verify everything is fine now and it looks good. I've realized it's a little annoying to run these manually to verify that the functionality is OK. Will have to think of a way to automate this.

closed 2023-11-26 by sagatowski