tcunit/TcUnit #212: TEST_FINISHED_NAMED no longer works on current master

ZLLentz, 2023-05-04:
**Describe the bug**
Tests that run properly using `TEST_FINISHED_NAMED` on version `1.2.0.0` do not run properly on the current master branch at commit 01461e89fc5d7fed14fd81611f9df9f4992ee0b8, which calls itself version `1.2.1.0`.

Instead, the following error is seen and we receive no further test output:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error		5/4/2023 8:35:27 AM 834 ms | 'Task 3' (350): Failed to find test 'AutoReset'				
```

**To Reproduce**
Create a test suite that finishes using `TEST_FINISHED_NAMED` instead of `TEST_FINISHED` and build and install the latest TcUnit master branch.
Here's the test suite this was discovered in today: https://github.com/pcdshub/lcls-twincat-general/blob/645da03cd87a19ce14b2181612f08c6cc8615937/LCLSGeneral/LCLSGeneral/POUs/Logger/Tests/FB_CircuitBreaker_Test.TcPOU#L5

**Expected behavior**
I expected the test to pass or fail as normal based on the assert results.

**Screenshots**
Sometimes it shows this error once, other times it shows the error twice. No other output from TcUnit is seen.
[image]

**Software versions**
Which version of the applicable software did you use? (TcUnit library, TwinCAT XAE version, Visual Studio version etc)
TcUnit master at 01461e89fc5d7fed14fd81611f9df9f4992ee0b8
TwinCAT XAE `4022.23`
TcXAE Shell `15.0.28307.1300 D15.9`

**Run environment**
Windows 10 Virtual Machine (on both `1.2.0.0` and `1.2.1.0`)

**Additional context**
I suspect these tests can be rewritten using `TEST_FINISHED` but I don't know for sure.

sagatowski, 2023-05-10:
Thanks for this very detailed report.
I have confirmed this is indeed an issue. I'll make sure it's fixed before the release of `1.3.0.0.`

ZLLentz, 2023-05-18:
I agree, our local test suite with `TEST_FINISHED_NAMED` now passes with the master branch version. I'll close this issue. Thank you!

closed 2023-05-18 by ZLLentz