tcunit/TcUnit #213: TcUnit.org website from source code

sagatowski, 2023-05-10:
The current website and documentation is a wordpress website. This doesn't play very well from a maintenance perspective as the only person that can maintain the website is me. Move the complete website into markdown-files as part of the repository, and create a GitHub action so that a website is automatically generated from source-code.
This will make maintenance of the documentation much easier, and when features are added, a requirement can be to also include documentation for new features.

sagatowski, 2023-11-19:
Implemented. TcUnit.org is not fully generated from github.com/tcunit/tcunit/docs.

closed 2023-11-19 by sagatowski

ahuca, 2023-12-08:
Hi, this might be related to this issue https://github.com/tcunit/TcUnit-Runner/issues/52.