tcunit/TcUnit #217: No test outputs shown in XAE shell when using TcBSD (simulated in VM)

paddenstoeltje, 2023-08-02:
I have setup a project with unit tests. For testing purposes I use a TcBSD image in a VM (using virtualbox). However, when running the tests in this environment, I do not see any feedback in the output window of the XAE shell (the tests have runned normally). When running the same project on a windows PC, the results of the test are displayed as expected. 

Is this because I'm using TcBSD or is it because of the VM? Anyone experiencing the same issue?

Running TC4024.47, with TcUnit 1.2

sagatowski, 2023-08-02:
Does it work if you run a VM with windows?

paddenstoeltje, 2023-08-03:
Hi Jacob,

I just gave it a try with a windows VM: this works like a charm, so probably TcUnit doesn't play nice with TcBSD?

Wouter

sagatowski, 2023-08-03:
Hi @paddenstoeltje!

And did you verify that the tests were indeed running on the TC/BSD machine? I mean, if you add a simple counter in the program, does it increment?

paddenstoeltje, 2023-08-04:
Hi Jacob,

I reinstalled the TcBSD; and now it works. No clue what went wrong, I assume somewhere a faulty setting of the VM / TcBSD.

Wouter

closed 2023-08-04 by paddenstoeltje