tcunit/TcUnit #222: Fails to execute TcUnit on TwinCAT user mode runtime

philippleidig, 2023-09-10:
**Describe the bug**
When running a TwinCAT XAE project on the new Beckhoff "[TC1700 | TwinCAT 3 Usermode Runtime](https://www.beckhoff.com/en-en/products/automation/twincat/tc1xxx-twincat-3-base/tc1700.html)" including a plc project with TcUnit as library reference some errors occur and the user mode runtime fails on startup.

The supposed cause may be the libraries used by 3S / CoDeSys. However, this would have to be investigated in more detail.

[image]

[image]

Are there any future plans for running TcUnit also on the "TwinCAT 3 Usermode Runtime"?

**To Reproduce**
Run a new XAE project inlcluding a plc project with TcUnit as library reference

**Software versions**
TcUnit v1.2.1.0
TwinCAT User Mode Runtime Build 4024.50

sagatowski, 2023-09-10:
Usermode runtime is not yet officially released, it will be included in 4026 so we have to wait until it's out of Beta-phase and see if this problems still persists.

I definately see many usecases for TcUnit running in Usermode runtime, as I've discussed in this video: https://youtu.be/mbQSRNMJRz8?si=XJx0e0JtT9l-Oubj

sagatowski, 2023-11-09:
I'll close this ticket until usermode runtime is released and is not a beta. If problems persists, we can re-open the ticket.

closed 2023-11-09 by sagatowski

philippleidig, 2023-11-11:
@sagatowski 

FIxed with TwinCAT >= 4024.53 :)

sagatowski, 2023-11-24:
@philippleidig Fantastic! 😊