tcunit/TcUnit #228: Duration not calculated for TEST_ORDERED() tests

dfreiberger, 2023-12-07:
**Describe the bug**
Tests defined used `TEST_ORDERED()` do not appear to calculate the duration correctly.
[image]

A quick look into the code reveals that the Test StartedAt may not be set, as shown in this screenshot for `TEST_FINISHED()`.
[image]

It looks like a simple fix to the `TEST_ORDERED` FC to add `Test.SetStartedAtIfNotSet(Timestamp := F_GetCpuCounterAs64bit(GVL_TcUnit.GetCpuCounter));` is necessary. I will try to submit a PR as soon as I have time for this.

**To Reproduce**
Run any set of tests that include `TEST_ORDERED` tests. For example `TestCase1C` will be calculated incorrectly while TestCase1D is correct.

```

IF TEST_ORDERED('TestCase1C') THEN

    AssertTrue(TRUE, 'Condition is not true');

    TEST_FINISHED();
END_IF

TEST('TestCase1D');

AssertTrue(FALSE, 'Condition is not true');
AssertTrue(TRUE, 'Condition is not true');

TEST_FINISHED();
```

**Expected behavior**
The duration should be calculated for the overall length of time that the test takes.

**Software versions**
1.3.0.0

**Run environment**
Hardware PLC

dfreiberger, 2023-12-07:
Easy fix in https://github.com/tcunit/TcUnit/commit/8fdf2564b54d48e4d64c646f0682e12227cb9d52 

I will look at adding unit tests tomorrow before creating PR.

sagatowski, 2023-12-07:
@dfreiberger Thanks for this discovery. Looking forward to your PR. ❤

Are you using [TcUnit-VsExtension](https://github.com/philippleidig/TcUnit-VsExtension) for your nice integration of TcUnit into TwinCAT?

Also, something feels different about the colors and font of your IDE? What nice font is that?

dfreiberger, 2023-12-08:
@sagatowski yes, I am using TcUnit-VsExtension + the adapter using the branch I am working on at [TcUnit-VsTestAdapter](https://github.com/dfreiberger/TcUnit-VsTestAdapter/tree/dfreiberger/parse-tmc) that can parse out tests via the TMC file and regex parsing for TEST and TEST_ORDERED sections. The fonts and colors I think are just defaults in TwinCAT VS 2019 although maybe I adjusted something at one point.
[image]
[image]

dfreiberger, 2023-12-21:
Fixed in PR #231

sagatowski, 2024-08-05:
Fixed in https://github.com/tcunit/TcUnit/pull/231 &

closed 2024-08-05 by sagatowski