tcunit/TcUnit #235: Tests seem to run 4 times

Oleksiy1990, 2024-02-01:
**Describe the bug**
When running the tests, it looks like the same output is produces 4 times

**To Reproduce**
What steps are necessary to reproduce the behavior?

**Expected behavior**
They should only run once and produce a single report message

**Screenshots**
[image]

**Software versions**
TcUnit 1.3, TwinCAT Build 4024.53

**Run environment**
Virtual (isolated core on the dev laptop)

**Additional context**
Add any other context about the problem here.

sagatowski, 2024-02-02:
This is not a problem with TcUnit.
Duplicate of https://github.com/tcunit/TcUnit/issues/92.

closed 2024-02-02 by sagatowski

Oleksiy1990, 2024-02-02:
Hi Jakob,

That's really nice info, thanks a lot!

Best regards,
Oleksiy