tcunit/TcUnit #264: Fails to execute TcUnit on TwinCAT 4026 user mode runtime

piratecarrot, 2025-09-30:
**Describe the bug**
When running a TwinCAT XAE project on Beckhoff "[TC1700 | TwinCAT 3 Usermode Runtime](https://www.beckhoff.com/en-en/products/automation/twincat/tc1xxx-twincat-3-base/tc1700.html)" including a PLC project with TcUnit as library reference some errors occur and the user mode runtime fails on startup.
Very similar to #222.

**To Reproduce**
Run a new XAE project including a PLC project.
Implemented program to ensure Usermode Runtime works, simple program with incrementing integer, everything works.
Then add TcUnit as library reference, cannot run program.

**Screenshots**
[image]

**Software versions**
TcUnit 1.3.1
TwinCAT 4026.18
Windows 10 Professional

**Run environment**
Windows 10 PC with Usermode Runtime

sagatowski, 2025-10-01:
It seems you have IO enabled. Make sure you disable IO before running TcUnit.

patgmiller, 2025-11-03:
This helped me https://github.com/tcunit/TcUnit/issues/264#issuecomment-3354837250, i had the exact same problem after adding the TcUnit library and the exact same error would occur when trying to activate with UmRT. Adding and setting the `HeapMemSizeMB` pair fixed it.

Thanks @piratecarrot and @sagatowski